OpenGarage/OpenGarage-Firmware #2: WeMos D1 Mini- Access point mode works great but not webserver mode

terrag42, 2016-04-25:
Today I tried flashing OpenGarage to a WeMos D1 Mini without any of the input / output hardware attached. I changed the defines.h file to reflect the [pin assignments ](http://www.wemos.cc/Products/d1_mini.html)of my board.
After it was done uploading, the blue light flashed rapidly, so I knew it was in access point mode. I connected to the access point and entered my credentials. It connected to my home network, and told me the IP address I should visit once I reconnect my device (my phone in this case) to the home network. Since the WeMos's D0 pin (GPIO 16) was not connected to the reset pin, I pressed the reset button and it started this strange flashing pattern with the blue led, two long and one short flash. When I ran the "Fing" app on my phone the ESP showed up on the assigned address, but when I went to ping it, it gave me over 90% packet loss. Could there be a problem since I'm using version 1.6.8 of the Arduino IDE and version 2.2.0 of the Arduino core (in boards manager)? Do I need to connect the ultrasonic range finder or the push button for it to work correctly?
Thanks for your help.

bepursuant, 2016-05-17:
I think you have to have the ultrasonic sensor connected, or it will WDT reset on you. At least, that's what I saw in my testing.... It basically sends the ping and waits for the response... until the wdt fires and restarts the module.

rayshobby, 2016-08-15:
The firmware indeed assumes ultrasonic distance sensor exists. Without it, reading the sensor would time out and cause watchdog timer reset. The easiest work-around is to find the read_distance function and return a fake value right away.

closed 2016-08-15 by rayshobby