OpenGarage/OpenGarage-Firmware #8: Change Connected SSID without a Reset?

HoECoder, 2017-07-06:
Maybe not an issue, but I was wondering if you'd be open to an additional setting for changing the SSID the device is connected to.

I recently made some changes to my home network and needed to swap SSID and passphrase, which meant climbing my ladder to the top of my garage in the summer heat to reset my OpenGarage.

I know it is fraught with some ugly edge cases (what if the network you are attempting to connect to rejects the connection).

But I think a valid use case could be:
1. Input new SSID and Passphrase
2. If connected, OK, proceeed.
3.a. If failed, move back to Open AP mode.
OR
3.b. Go back to the previous SSID and passphrase.

I don't know how close to the memory edge the device is currently and if it could support such a thing.

Thoughts?

BlueArcherX, 2017-07-08:
Alternatively, a reset to defaults button in the web UI

rayshobby, 2017-07-16:
This will be addressed in the upcoming 1.0.6 firmware (source code changes already committed in the repository, compiled firmware will be released soon).

closed 2017-07-16 by rayshobby

HoECoder, 2017-07-17:
Awesome! Thanks again, Ray!