OpenGarage/OpenGarage-Firmware #9: Add MQTT

DynamoBen, 2017-07-16:
Incorporate MQTT into master.

See here: https://opengarage.io/forums/topic/mqtt-and-og/

rayshobby, 2017-11-30:
MQTT will be supported in the upcoming firmware 1.0.7 (merging Jeff Lawrence's contributions).

closed 2017-11-30 by rayshobby

xbmcnut, 2018-05-18:
Running og_1.0.8.bin with MQTT connected and device configured as a cover in home assistant. I want to be able to open and close the garage door via MQTT which is throitically supported? Can anyone advise the publish topic and payload? I tried subscribing to GarageDoor/# and get nothing.

DynamoBen, 2018-05-18:
MQTT control isn't implemented, only status updates. It could be added, you might want to create a new feature request ticket for that.

xbmcnut, 2018-05-23:
Will do, thanks.