OpenGarage/OpenGarage-Firmware #11: sensor keeps opening/closing & signal beeping

jjanderson, 2017-10-19:
Hi, I have the latest firmware. I also have the Home assistant component and it all works well except a couple of issues:

1. Keep getting notifaction that the door is open in both Blynk and Home assistant. I have a ceiling mount and threshold set a at 40... This is annoying as I cannot use this for notifications at present as these go off all the time. I have attached a screenshot of the sensor. Red bars indicate that the door is open.
[image]

2. I use the Linksys Velop wifi mesh. The unit is placed in the Garage (duh ;-) which picks up 2 of the velop AP's. The issue is that the unit then keeps beeping every time it disconnects and reconnects... Is there anyway I can turn this off?

Thanks

codahq, 2017-12-17:
To reproduce issue number one (this issue really should have been submitted as one issue and one enhancement request) all you need to do is get an app or decide that does an entire subnet scan or discovery. For example, download Fing from an app store and use it to discover devices. When the opengarage is pinged or scanned it tends to affect the sensor and it will trigger open notifications if they are configured. Then it will immediately correct after the scan is over and send the closed notification since the for really is closed.

This is particularly annoying because I have a device that is very network chatty with multicast packets maybe and I always get garage notifications when I turn the chatty device on or off. 

I haven't looked at the open garage firmware code to see why it does this but it's definitely a firmware bug this is always reproducible. Please fix this.

mattie47, 2018-01-21:
@jjanderson 

Would you mind sharing what you did to get this working in Home Assistant? I've been currently trying to do this, and any steps/examples would be greatly appreciated.

Thanks

mattie47, 2018-01-21:
Ahhh ignore. I've only just noticed there is a component for this:

https://home-assistant.io/components/cover.opengarage/

closed 2018-01-26 by rayshobby

HITChris, 2020-01-08:
@codahq Can you clarify what the fix/workaround is? New user/hardware, and I'm seeing the same, consistent open/close door state cycles that @jjanderson described (for which you seemed to imply a firmware issue is at hand).

codahq, 2020-01-08:
I don't remember what the fix/workaround is.  I forked the project and added SmartThings/Hubitat support and I probably fixed this bug or worked around it in code.  I'm too busy to look or worry about this project right now though.

jjanderson, 2020-01-08:
Make sure you have the latest firmware... that initial bug was almost 2.5 years ago and thus a couple of firmwares ago... i have ZERO issues with this at the moment...

HITChris, 2020-01-09:
I'm on release 1.1.0, but I'll give the dev branch a shot. Perhaps I'm just seeing a different behavior, and/or my hardware is at fault. Thanks!

jjanderson, 2020-01-09:
Maybe try reflash the firmware?

 

Ray is a really helpful guy, so maybe also try and reach out too hm?

codahq, 2020-01-09:
Are you able to reliably reproduce this?  I was able to at any time I wanted on any version of this firmware with the steps I provided above.  Just download the "Fing" app and run a scan.  I'm almost 100% positive that certain "smart home" products from Ring/Google will cause this issue to occur as well (on certain events because they tend to flood the network with packets).  

I haven't looked at the firmware code in a long while but from what I remember I thought under normal circumstances it wouldn't be hitting the lines of code that sent the notifications.  Instead I decided that something was either getting overflowed and memory was getting stepped on or the device was crashing/rebooting and the notification was being sent when it recovered and came back up.  

Troubleshooting your hardware and flashing again probably won't help.  I did all of that.  I'm almost 100% certain it's a bug in the firmware in this code or in one of the libraries it uses.  Ray saw this bug and closed it without a comment (which bugs me a little).  Maybe reaching out via email will help but I wouldn't get your hopes up too high.  He probably gets these comments as emails because he's subscribed to this repository.  I'm not going to look into it further because it isn't a priority for me anymore since I moved away from Blynk forever ago and I use a custom firmware and a different method for notifications.  

To be completely frank I thought this firmware was sort of a mess.  If somehow I finished all of my other projects the very last thing I would do is rewrite it from scratch or move to Tasmota and use the raw sensor values and relay in automation rules.