OpenGarage/OpenGarage-Firmware #16: 1.0.7 doesn't stick

xbmcnut, 2018-02-28:
Hi there,

Purchased a device 6-8 months ago and only just sparked it up for the 1st time today. Came with 1.0.6 and I tried updating to 1.0.7 and it won't work with that firmware. Do I need newer hardware for that version? I used the og_reflash to roll it back to 1.0.6 and immediately give me the 4 tones after flashing, something 1.0.7 doesn't do.

rayshobby, 2018-02-28:
Firmware 1.0.7 has removed the start-up sound. It's likely that when you upgrade to 1.0.7 the controller will revert back to AP mode (because 1.0.7 is compiled with a different ESP8266 core, so the flash layout has changed).

xbmcnut, 2018-02-28:
Nope, would not work at all with 1.0.7 however I was here https://github.com/OpenGarage/OpenGarage-Firmware/releases instead of here https://github.com/OpenGarage/OpenGarage-Firmware/tree/master/Compiled so after installing 1.0.8, everything is working again thank you.

On another note, is it possible to connect a door reed to this model I have? What pins do I use?

xbmcnut, 2018-02-28:
Found instructions for integrating an external reed switch here https://github.com/lawrence-jeff/OpenGarage-Firmware#using-magnetic-switch.

closed 2018-02-28 by xbmcnut