OpenGarage/OpenGarage-Firmware #18: Feature Request: Temperature Sensor Support

mattie47, 2018-03-07:
I really love this project, and it's amazing how polished it is :-) 

 

I've been using this for the last month and have it integrated with Home Assistant along with Home Bridge (for Siri support). 

[image]

More on getting that working here: https://community.home-assistant.io/t/opengarage-open-source-wifi-garage-opener/17228/112?u=mattie47

I've been trying to get a DHT22 sensor to also work, and hopefully report the data via MQTT back to Home Assistant, and possibly report it in the Web interface. 

I've tried to follow the instructions in this thread https://opengarage.io/forums/topic/wemos-d1-open-garage-with-dht22-temperaturehumidity-sensor/ however haven't had much luck (and have replied there also).

I'm just curious if there are any plans to add temp support, and or whether there may already be a fork that perhaps have temp support in it? :-) 

Thanks,

Matt

jjlawren, 2018-03-08:
It probably needs to be made more generic so I'm not making it a pull request, but this works for me:

https://github.com/jjlawren/OpenGarage-Firmware/commit/e2220d754f4502b996f8e6966dcf2f2685df6fb1

mattie47, 2018-03-16:
@jjlawren  Thanks for that. Much appreciated 

 I have had a go with it but still haven't got it working yet. I think my DHT22 is broken (and I only have one). I tried with a dht11 and got weird output (1000 or 990 degrees). And yes I did make sure to change type to dht11). 

Anyway - When  I get some time I'm going to look into this further and update here. 

Thanks,

Matt

rayshobby, 2018-06-09:
I've been using the AM2320 T/H sensor recently and I like it better than DHT11/DHT22. Because AM2320 uses I2C, it can share the same I2C bus with other I2C devices and won't occupy additional GPIOs.

mattie47, 2018-06-11:
Interesting. I hadn't heard of the AM2320. I did eventually get this working, but I can't remember what my problem was. I think I had one of the arduino IDE packages in the wrong location or something, and changes I was making weren't actually getting pulled into the build. 

So I've been using @jjlawren's code for probably 3 months.  Occasionally I stop getting temp readings sent back to my mqtt server, but I've had it been quite ruthless sending stats I think every 5 or 10 seconds.

jakesza, 2018-11-06:
Did anybody get the DHT sensors to work on 1.0.9? I tried [https://github.com/OpenGarage/OpenGarage-Firmware/issues/18#issuecomment-371530590](url) but no MQTT messages.

rayshobby, 2018-12-08:
Are you referring to DHT sensor, or MQTT? The link you referred to is invalid.

rayshobby, 2019-02-03:
Temperature/humidity sensor support has been added to firmware 1.1.0:
https://opengarage.io/forums/topic/og-firmware-1-1-0-released/
although it hasn't been added to MQTT yet, it's available through the HTTP API and we will add to MQTT as well as soon as the feature has been tested out.

closed 2019-02-03 by rayshobby

AlBrough, 2020-02-14:
Could T & H please also be added from Blynnk integration?

sjamthe, 2020-02-14:
+1 for blynk charts for both. will be very helpful.

thanks