OpenGarage/OpenGarage-Firmware #21: Failed compilation with makeESPArduino

ravishivt, 2018-03-19:
I get the following errors when compiling with makeESPArduino.  Here are the tags/versions of each required package I am using:

makeESPArduino - 4.11.0
blynk-library - v0.5.1
pubsubclient - v1.99.2
OpenGarage - 1.0.7

```
root@1eede81c43bf:/usr/local/src/og/OpenGarage# make
uart.c
time.c
Tone.cpp
core_esp8266_wiring_digital.c
core_esp8266_flash_utils.c
pgmspace.cpp
umm_malloc.c
core_esp8266_si2c.c
core_esp8266_wiring_shift.c
core_esp8266_postmortem.c
Schedule.cpp
cbuf.cpp
cont_util.c
base64.cpp
core_esp8266_wiring_analog.c
spiffs_api.cpp
core_esp8266_wiring_pwm.c
libc_replacements.c
core_esp8266_phy.c
StreamString.cpp
core_esp8266_wiring_pulse.c
WMath.cpp
Esp.cpp
HardwareSerial.cpp
WString.cpp
core_esp8266_i2s.c
core_esp8266_timer.c
abi.cpp
core_esp8266_wiring.c
core_esp8266_eboot_command.c
Stream.cpp
MD5Builder.cpp
core_esp8266_main.cpp
debug.cpp
IPAddress.cpp
spiffs_check.c
cdecode.c
spiffs_gc.c
spiffs_nucleus.c
spiffs_hydrogen.c
core_esp8266_noniso.c
spiffs_cache.c
cencode.c
heap.c
Print.cpp
ogMainArduino.ino
cont.S
ESP8266WiFiScan.cpp
WiFiUdp.cpp
ESP8266WiFiSTA.cpp
Wire.cpp
ESP8266WiFi.cpp
ESP8266WiFiMulti.cpp
WiFiClient.cpp
ESP8266WiFiGeneric.cpp
WiFiClientSecure.cpp
WiFiServer.cpp
ESP8266WiFiAP.cpp
ESP8266mDNS.cpp
DNSServer.cpp
BlynkDebug.cpp
[... 1043 lines cut ...]
    PubSubClient(Client& c, String hostname, uint16_t port = 1883);
    ^
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:89:4: note:   candidate expects 3 arguments, 4 provided
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:87:4: note: PubSubClient::PubSubClient(Client&, IPAddress&, uint16_t)
    PubSubClient(Client& c, IPAddress &ip, uint16_t port = 1883);
    ^
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:87:4: note:   candidate expects 3 arguments, 4 provided
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:84:4: note: PubSubClient::PubSubClient(Client&)
    PubSubClient(Client& c);
    ^
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:84:4: note:   candidate expects 1 argument, 4 provided
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:20:7: note: PubSubClient::PubSubClient(const PubSubClient&)
 class PubSubClient {
       ^
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:20:7: note:   candidate expects 1 argument, 4 provided
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:20:7: note: PubSubClient::PubSubClient(PubSubClient&&)
/usr/local/src/esp8266-arduino/libraries/pubsubclient/src/PubSubClient.h:20:7: note:   candidate expects 1 argument, 4 provided
Makefile:236: recipe for target '/tmp/ogMainArduino/ogMainArduino_generic/publish_spec.cpp.o' failed
make: *** [/tmp/ogMainArduino/ogMainArduino_generic/publish_spec.cpp.o] Error 1
Makefile:236: recipe for target '/tmp/ogMainArduino/ogMainArduino_generic/connect_spec.cpp.o' failed
make: *** [/tmp/ogMainArduino/ogMainArduino_generic/connect_spec.cpp.o] Error 1
Makefile:236: recipe for target '/tmp/ogMainArduino/ogMainArduino_generic/receive_spec.cpp.o' failed
make: *** [/tmp/ogMainArduino/ogMainArduino_generic/receive_spec.cpp.o] Error 1
root@1eede81c43bf:/usr/local/src/og/OpenGarage#
```

rayshobby, 2018-06-09:
I have no idea why you are getting these errors. What's your ESP8266 core version?

closed 2018-09-10 by rayshobby

LeoLaiFookSiang, 2019-02-08:
I am facing this issue too, may I know how you solve it ?