OpenGarage/OpenGarage-Firmware #23: Auto close not always working.

identd113, 2018-04-15:
I currently have the auto close feature setup, and working, but sometimes it takes much longer then the 20 minutes configured to close the garage.  This is more typical when using the RF remote to open the garage.  This was not occurring in the previous 1.0.6 release.  I am currently running 1.0.8 firmware/

I have included the open and close log below:

```
Event | DateTime | Details
-- | -- | --
Closed | 4/15/2018, 7:42:51 AM | 97 cm
Opened | 4/15/2018, 7:41:09 AM | 12 cm
Closed | 4/14/2018, 5:52:48 PM | 450 cm
Opened | 4/14/2018, 4:48:30 PM | 13 cm
Closed | 4/14/2018, 4:48:21 PM | 264 cm
Opened | 4/14/2018, 4:40:39 PM | 13 cm
Closed | 4/14/2018, 4:39:45 PM | 263 cm
Opened | 4/14/2018, 4:29:30 PM | 12 cm
Closed | 4/14/2018, 8:56:21 AM | 264 cm
Opened | 4/14/2018, 8:36:03 AM | 13 cm
Closed | 4/14/2018, 6:36:45 AM | 262 cm
Opened | 4/14/2018, 6:35:45 AM | 13 cm
Closed | 4/14/2018, 12:23:12 AM | 90 cm
Opened | 4/13/2018, 11:35:06 PM | 13 cm
Closed | 4/13/2018, 11:01:06 PM | 98 cm
Opened | 4/13/2018, 10:39:12 PM | 13 cm
Closed | 4/13/2018, 9:40:30 PM | 217 cm
Opened | 4/13/2018, 9:34:54 PM | 13 cm
Closed | 4/13/2018, 3:23:18 PM | 261 cm
Opened | 4/13/2018, 2:14:27 PM | 13 cm
Closed | 4/13/2018, 12:24:54 PM | 262 cm
Opened | 4/13/2018, 12:21:36 PM | 13 cm
Closed | 4/13/2018, 11:54:09 AM | 86 cm
Opened | 4/13/2018, 10:10:06 AM | 13 cm
Closed | 4/13/2018, 12:10:51 AM | 87 cm
Opened | 4/12/2018, 10:07:27 PM | 13 cm
Closed | 4/12/2018, 7:36:39 PM | 85 cm
Opened | 4/12/2018, 6:49:00 PM | 13 cm
Closed | 4/12/2018, 1:13:33 PM | 85 cm
Opened | 4/12/2018, 12:53:12 PM | 12 cm
Closed | 4/12/2018, 9:45:21 AM | 261 cm
Opened | 4/12/2018, 9:42:15 AM | 12 cm
Closed | 4/12/2018, 7:23:42 AM | 88 cm
Opened | 4/12/2018, 7:21:12 AM | 13 cm
Closed | 4/11/2018, 10:15:48 PM | 87 cm
Opened | 4/11/2018, 10:15:12 PM | 13 cm
Closed | 4/11/2018, 10:14:33 PM | 87 cm
Opened | 4/11/2018, 10:13:06 PM | 13 cm
Closed | 4/11/2018, 7:31:12 PM | 88 cm
Opened | 4/11/2018, 5:39:46 PM | 12 cm
Closed | 4/11/2018, 4:57:01 PM | 260 cm
Opened | 4/11/2018, 4:55:55 PM | 12 cm
Closed | 4/11/2018, 4:47:22 PM | 85 cm
Opened | 4/11/2018, 4:27:01 PM | 12 cm
Closed | 4/11/2018, 7:48:07 AM | 263 cm
Opened | 4/11/2018, 7:45:43 AM | 13 cm
Closed | 4/11/2018, 6:34:52 AM | 262 cm
Opened | 4/11/2018, 6:33:01 AM | 13 cm
Closed | 4/10/2018, 7:22:55 PM | 86 cm
Opened | 4/10/2018, 6:53:37 PM | 12 cm
Closed | 4/10/2018, 6:48:34 PM | 86 cm
Opened | 4/10/2018, 6:13:28 PM | 12 cm
Closed | 4/10/2018, 4:50:58 PM | 262 cm
Opened | 4/10/2018, 4:45:25 PM | 13 cm
Closed | 4/10/2018, 7:54:43 AM | 86 cm
Opened | 4/10/2018, 7:52:16 AM | 13 cm
Closed | 4/9/2018, 10:36:40 PM | 87 cm
Opened | 4/9/2018, 9:13:58 PM | 13 cm
Closed | 4/9/2018, 8:20:37 PM | 262 cm
Opened | 4/9/2018, 5:59:10 PM | 13 cm
Closed | 4/9/2018, 12:23:04 PM | 262 cm
Opened | 4/9/2018, 12:16:07 PM | 13 cm
Closed | 4/9/2018, 8:43:07 AM | 87 cm
Opened | 4/9/2018, 8:41:55 AM | 13 cm
Closed | 4/9/2018, 8:22:19 AM | 265 cm
Opened | 4/9/2018, 8:10:34 AM | 12 cm
Closed | 4/9/2018, 6:17:22 AM | 88 cm
Opened | 4/9/2018, 6:12:46 AM | 13 cm
Closed | 4/8/2018, 6:41:13 PM | 88 cm
Opened | 4/8/2018, 5:55:40 PM | 13 cm
Closed | 4/8/2018, 7:43:37 AM | 89 cm
Opened | 4/8/2018, 7:31:52 AM | 12 cm
Closed | 4/7/2018, 9:29:55 PM | 88 cm
Opened | 4/7/2018, 9:28:13 PM | 13 cm
Closed | 4/7/2018, 5:55:52 PM | 262 cm
Opened | 4/7/2018, 5:28:43 PM | 12 cm
Closed | 4/7/2018, 12:21:55 PM | 265 cm
Opened | 4/7/2018, 12:20:43 PM | 13 cm
Closed | 4/7/2018, 11:49:55 AM | 90 cm
Opened | 4/7/2018, 11:41:22 AM | 13 cm
Closed | 4/7/2018, 11:23:22 AM | 89 cm
Opened | 4/7/2018, 11:20:52 AM | 13 cm
Closed | 4/6/2018, 7:54:46 PM | 90 cm
Opened | 4/6/2018, 6:40:20 PM | 13 cm
Closed | 4/6/2018, 5:34:53 PM | 89 cm
Opened | 4/6/2018, 5:29:02 PM | 12 cm
Closed | 4/6/2018, 4:31:41 PM | 90 cm
Opened | 4/6/2018, 4:28:20 PM | 13 cm
Closed | 4/6/2018, 4:02:23 PM | 89 cm
Opened | 4/6/2018, 4:00:53 PM | 13 cm
Closed | 4/6/2018, 2:11:59 PM | 264 cm
Opened | 4/6/2018, 1:35:47 PM | 13 cm
Closed | 4/6/2018, 6:36:23 AM | 283 cm
Opened | 4/6/2018, 6:35:32 AM | 13 cm
Closed | 4/5/2018, 7:49:53 PM | 88 cm
Opened | 4/5/2018, 5:56:17 PM | 12 cm
Closed | 4/5/2018, 12:47:02 PM | 88 cm
Closed | 4/5/2018, 9:53:20 AM | 265 cm
Opened | 4/5/2018, 9:46:23 AM | 13 cm
```

rayshobby, 2018-06-09:
What's the time limit you set for "If open for longer than: xxx minutes"?

identd113, 2018-06-09:
20 minutes

rayshobby, 2018-06-09:
If it's set to 20 minutes then it will take roughly 20 minutes (or maybe up to 21 minutes) for the auto-close to kick in. You posted a fairly long log, are there some particular data entries in the log that indicate a problem?

identd113, 2018-06-09:
I am seeing much longer delays in some cases. I think the log shows delays up to 60 minutes or longer. I will test this right now and see if I can replicate

identd113, 2018-06-09:
[image]

identd113, 2018-06-09:
It has been about 25 minutes, and it is still open.
6/8 entry shows 74 minute gap in the close

rayshobby, 2018-06-09:
There are two potential source of issues and we need to first figure out what was the cause. The first is whether the controller detects the timing correctly: if it does, your Blynk app should receive a notification about the door being open for longer than 20 minutes and it will attempt to trigger a close. If you receive that notification around 20 minutes after the door is open, that means the timing is correct.

The second issue is that it may attempt to close but for some reason cannot close it. That could be due to a variety of reasons, such as an external service (e.g. IFTTT) that's keeping the Blynk button pressed. In any case, if you can tell me whether you receive the Blynk notification in the first part, it will help understand which cause it is.

identd113, 2018-06-09:
I do not have blynk or ifttt setup currently

identd113, 2018-06-09:
This did automatically close at 1:03 which was 73 minutes

identd113, 2018-06-09:
Interestingly it seems like it’s almost always 73 minutes ( give or take 1 minute). I re opened the garage at 2:37 and based on this would expect it to close at 3:50. I will send a screenshot of the log if this occurs

identd113, 2018-06-10:
OK, so it was over 20 minutes, but not 73
[image]

identd113, 2018-10-26:
I am still seeing this. How is the time element handled internally?  It seems epoch or 8601 would be the way to go

rayshobby, 2018-10-29:
It represents time in epoch time. You can check the source code yourself:
https://github.com/OpenGarage/OpenGarage-Firmware/blob/master/OpenGarage/main.cpp#L898
Honestly you are the only person reporting this issue so far. I'm pressure sure this is not a software issue. Could be a hardware issue.

rayshobby, 2018-12-08:
Unless if there is further update, I will close this issue for now.

closed 2018-12-08 by rayshobby

mrblack7, 2019-02-12:
I am having the same issue as this.

I have it set to auto-close after 50 mins. Most of the time, it is not closing even after 1 hour... (see 
attached latest log as an example)

[image]
[image]

On a couple of occasions, I have had it auto-close within 10 mins. One time I was backing out the car and it nearly closed the door on me!

I didn't have this issue prior to updating 1.0.9  a couple of weeks ago. I was previously on 1.0.6 I think.

Let me know if you need any more info. Happy to help!

mrblack7, 2019-02-12:
Just had another instance of the door closing automatically within a few minutes....

[image]