OpenGarage/OpenGarage-Firmware #24: Feature Request: Add ability to toggle door via MQTT

xbmcnut, 2018-05-23:
Have one deployed so thank you and working well with Home Assistant. Would be awesome to be able to control the door with a MQTT payload.

rayshobby, 2018-06-09:
This has already been implemented:
https://github.com/OpenGarage/OpenGarage-Firmware/blob/master/OpenGarage/main.cpp#L565
basically, if you publish a message to the MQTT server with any topic and message of 'Button', or the more recommended way is to publish to topic 'OGNAME/IN/STATE' with a message of 'close' or 'open', it will trigger the door action. OGNAME is your OpenGarage device's name (you can change it in Settings).

closed 2018-06-09 by rayshobby

xbmcnut, 2018-06-10:
Thanks, but does not seem to work. My door name is "PoakaGarage" so I'm publishing either close, open or Button to `PoakaGarage/IN/STATE` and nothing happens. Can open and close from Home Assistant OK.

xbmcnut, 2018-07-14:
Anyone tested this and confirm it is working?