OpenGarage/OpenGarage-Firmware #29: Unneccessary modification of Updater.h in ESP8266 library

kodaman2, 2018-08-11:
I am all for modifying existing libraries when there's a clear purpose. I don't believe we need to expose reset method at all.

The only place I see the reset used is here:

```
void on_sta_upload_fin() {

  if(!verify_device_key()) {
    server_send_result(HTML_UNAUTHORIZED);
    Update.reset();       // --> To this Update.end(false);
    return;
  }
```

This is an if statement from bool UpdaterClass::end(bool evenIfRemaining) which can be used to call reset if the upload is not finished, and we give it a false parameter.
```
if(hasError() || (!isFinished() && !evenIfRemaining)){
#ifdef DEBUG_UPDATER
    DEBUG_UPDATER.printf("premature end: res:%u, pos:%u/%u\n", getError(), progress(), _size);
#endif

    _reset();
    return false;
  }
```
Did I misinterpreted something along the way?

rayshobby, 2018-09-10:
OK, good suggestion, will merge the PR soon.

kodaman2, 2018-09-13:
Awesome got side tracked for a while but I am also working on updating html for gtm zones and saving light time, and options to choose units I believe that's the icing on the cake for opengarage. The whole project is awesome by the way. In the making is also an app to auto/close garage when car is approaching to the house but that's on hold for now super busy at work.

[image]

[image]