OpenGarage/OpenGarage-Firmware #32: OG 1.0.9 firmware gives blank page on Options

prophead67, 2019-01-01:
After upgrading my OG with the new 1.0.9 firmware, I noticed that clicking the options button in the web UI almost always results in a blank page at http://devip/vo. After rebooting the device a couple of times I can finally get the options page to display.

I've replicated this in Chrome and Internet Explorer. I can provide any logs/data you would like.

Thanks,
B.

jmyler, 2019-01-01:
Did the 1.0.9 get pulled? I don’t see it in releases anywhere.

rayshobby, 2019-01-03:
Take a look at this blog post and see if the re-compiled firmware fixes the issue:
https://opengarage.io/forums/topic/missing-options-page/#post-1376

prophead67, 2019-01-03:
Thank you Ray. Since this will reset my OG I will try it on Friday and
report back the results.

Cheers!

prophead67, 2019-01-05:
Greetings Ray, I uploaded the new firmware you linked but couldn't save my
old config due to the options screen being inaccessible. Went through the
whole AP mode setup and reset all my settings and integrations, as well as
generated a new Blynk token in the app to get connected. After all that,
the device reverted back to a blank options screen. It was accessible for a
short time and now it's blank again. Please see screenshots below.

[image: image.png]
[image: image.png]

rayshobby, 2019-01-06:
Is your OpenGarage purchased from us? If so, you can submit a support ticket and send it back to us so we can take a look. I can't reproduce this issue on our OpenGarage units.

rayshobby, 2019-01-08:
That't fine, you can still submit a support ticket with your Amazon order number, and we can arrange for service.

prophead67, 2019-01-09:
Thank you, I'll do that and go from there.

Cheers!

edwork, 2019-01-25:
Hey,

I just stumbled upon this thread but I'm also rolling OG 1.0.9 and ever since after a few days of use the WebUI and MQTT quit responding. The HTTP API keeps working however and I am able to reboot it by calling `http://172.21.6.20/cc?dkey=example&reboot=1` and after a few seconds everything is back to normal. I don't need access to the webUI often, but it was MQTT that tipped me off. Auto Close continues to work but polling for attributes also quits working. 

I may end up reverting to 1.0.8 in the meantime to rule out a hardware issue (unit is only 1 year old). I'll look through the changes and see if I can find any breaking changes. I imagine that people who use IFTTT/Blynk might not be aware of the issue if they keep working normally and never need to access the webUI.

brianhks, 2019-01-25:
I just received my unit from you guys today and tried it out.  My options screen is blank.  I checked http headers and I get back a 200 but there is zero length content.

rayshobby, 2019-01-25:
@brianhks: please submit a support ticket so we can arrange for it to be shipped back to us to check. I just sample tested 5 units and all of them are fine, with no missing options page. So we need to get back your unit to take a look and debug.

rayshobby, 2019-02-03:
I've just pushed out firmware 1.1.0, which attempts to address the options page issue:
https://opengarage.io/forums/topic/og-firmware-1-1-0-released/
The main change is to directly send program memory strings instead of loading the entire string to RAM first, which hopefully will reduce RAM usage and likely eliminate the problem. Give it a try and see if it makes any difference.

prophead67, 2019-02-08:
I've also updated firmware from 1.0.9 to 1.1.0 and so far, so good. I will note that I did not have to start with a clean slate on the configuration, but I did have to reset most of the options on the Integration page including tokens and automation. I did not need to connect in AP mode or reset the network settings.

Nice touch adding a T/H sensor option! Are there any docs on the GPIO pins and their location? I've not cracked my device open before. Is it a standard ESP board?

Thanks for your help!
Cheers

prophead67, 2019-02-08:
Oops, sorry for the erroneous close!

edwork, 2019-02-16:
I've had my OG powered on for a little over a week now. I can still get valid responses from the REST API, however loading the webUI gives me (see attached images). As a side note, upgrading from 1.0.8 to 1.1.0 gave me a factory reset.  

[image]
[image]

prophead67, 2019-02-16:
Thanks for reminding me to check my OG, Ed. I didn't grab a screen capture
of it, but upon accessing it today I found a similar screen on the OG home
page. Refreshing the page got it back to normal. After that, all the
configuration pages appeared to be normal.

rayshobby, 2019-07-05:
This issue is solved (I assume) by firmware 1.1.0 (I haven't heard of this issue again since we started shipping OG with firmware 1.1.0 loaded).

closed 2019-07-05 by rayshobby