OpenGarage/OpenGarage-Firmware #34: Auto-close not working as expected

mrblack7, 2019-02-17:
(Creating a new issue for this)

I am having the same issue as this.

I have it set to auto-close after 50 mins. Most of the time, it is not closing even after 1 hour... (see 
attached latest log as an example)
The latest closed event was done manually using the RF remote.

[image]
[image]

On a couple of occasions, I have had it auto-close within 10 mins. One time I was backing out the car and it nearly closed the door on me!

I didn't have this issue prior to updating 1.0.9  a couple of weeks ago. I was previously on 1.0.6 I think.

Let me know if you need any more info. Happy to help!

_Originally posted by @mrblack7 in https://github.com/OpenGarage/OpenGarage-Firmware/issues/23#issuecomment-462611179_

mrblack7, 2019-02-17:
https://github.com/OpenGarage/OpenGarage-Firmware/issues/23#issuecomment-462898374

rayshobby, 2021-01-30:
Sorry, I can't reproduce this issue and I am not aware of this being a common problem.

closed 2021-01-30 by rayshobby