OpenGarage/OpenGarage-Firmware #35: F/W 1.10 unresponsive after 24 hours

callifo, 2019-03-11:
Updated to F/W 1.10 on my OpenGarage units about 48 hours ago, and twice now within 24 hours of each reboot, it has stopped responding via WiFi. I have two units, and they usually stop working within close proximity of each other. 

According to my WiFi access point, it is still connected, and sitting on the lowest data rate, but it cannot be pinged. Their webpage is not accessible, and nor does the API work. They are still flashing indicating they are polling. 

I use them with HomeAssistant which polls through the API. I do not use the Blynk or IFTTT integration.

callifo, 2019-03-12:
Rolled back to 1.0.9 and it is now doing it too. Was there anything changed in 1.10 that might be persisting when rolling back to 1.0.9?

These have been rock solid for over a year previous to the latest update.

hmbarbosa, 2019-03-12:
Same happened with me..

ShaunAndersonAZ, 2019-04-02:
I had the same thing happen too.  Updated yesterday and it's been unresponsive since.

callifo, 2019-04-25:
I tried rolling back to 1.0.8, but one of my two units was still doing the unresponsive behaviour. I used the factory reset, and that completely bricked it unfortunately. Unit wont even start anymore, just flashes the blue LED once, and stops.

rayshobby, 2019-07-05:
Per release notes: if you update from 1.0.9 to 1.1.0, the controller recovers to AP mode (because 1.1.0 changed the layout of the flash memory). Just follow the WiFi setup steps to re-enter WiFi SSID and password.

davidjb, 2019-09-07:
I'm on v1.1.1 (version pre-installed on purchase) and have seen the same effect as @callifo. 

The OG worked fine for most of a day, turned on & setup in the morning, but then stopped responding via WiFi in the evening (~7 hours later).  Like @callifo, it's reported as connected to my WiFi AP at around 24-48Mbps, so it _should_ be a solid connection.   Restarting the access point's Wifi sees the OG reconnect automatically and start responding again. 

Thoughts, @rayshobby?

callifo, 2020-01-02:
Just an update to this, I ended up moving to using MQTT as a test and this instability issue seems to have gone away. Previously Homeassistant was polling the device through HTTP every few seconds, and then after about 12-24 hours it would become unresponsive. Since changing to MQTT it hasn't happened again. 

It wasn't ideal as I couldn't monitor car present/away state, but gabe565 's beta 1.2.0 fork F/W had the MQTT distance messages present so I can calculate car in/out myself, and this was the best compromise.

psmedley, 2020-01-02:
FWIW - I'm polling the opengarage via http every 60 seconds from Openhab, and am not seeing any issues like this.

callifo, 2020-01-04:
> 
> 
> FWIW - I'm polling the opengarage via http every 60 seconds from Openhab, and am not seeing any issues like this.

Can you up it to once every 5 seconds and let it run for a few days? It would be nice to confirm whether its the timing or the way HA is doing the polling that's causing it.

psmedley, 2020-01-04:
Done, refresh set to 5 seconds as of 6pm Saturday Adelaide time.

psmedley, 2020-01-06:
over 48 hours now and the OG is still responding

rayshobby, 2020-07-28:
With firmware 1.1.1 I don't see this as a known/common problem, so I am closing this issue for now.

closed 2020-07-28 by rayshobby