OpenGarage/OpenGarage-Firmware #36: Web unresponsive after configuring mqtt

oedo808, 2019-04-09:
When I point to my Mosquitto server my webui becomes unresponsive.  I've tried this twice now, configure wifi, point to MQTT server, and then the webui will not come up.  As I'm typing this I just received two messages with the status, so I suppose it's still working I'm just unable to make any other configuration changes.

oedo808, 2019-04-17:
I forgot to mention that this device shipped with firmware 1.1.0.  I tried reapplying the firmware OTA which did not help.  I downgraded to 1.0.9 and it is fully functional.

apop880, 2019-07-24:
@oedo808 Does your MQTT broker require authentication? If so, I ran into the same issue. The MQTT integration doesn't fail gracefully if it runs into an issue. When I couldn't connect to my broker because it wanted a username and password, the OpenGarage HTTP interface went down as well. I struggled with it for awhile before turning off authentication on my broker, and then all was well.

It looks like the function that connects to MQTT never has a timeout attached to it, so it'll keep trying and never get up. And it's probably hitting that function so frequently that it doesn't have the resources to serve the web interface.

Authentication and failing gracefully are definitely two things the MQTT integration could use. I might take it upon myself to play with the code and see if I can add these things if I have time.

irqnet, 2019-07-26:
https://github.com/OpenGarage/OpenGarage-Firmware/issues/33

Got the same, posted on February as issue - was closed because user/pw connect to mqtt broker is unsupported.

nuff said

oedo808, 2019-07-26:
Actually my broker is mosquito and it's unauthenticated. I rolled back to a
previous firmware, maybe 1.10, I don't remember and it worked fine.
Haven't touched it since but I will need to eventually.  Receiving commands
from the opener is close enough to instant, but sensor notifications for
open/close are sometimes quick but usually a couple of minutes late and
frequently 10 minutes late when opened with a standard garage door opener.
It seems like the device goes to sleep and only wakes up on a schedule.

apop880, 2019-07-26:
Interesting, it should be continuing to transmit the status at regular intervals. You might want to use an application such as MQTT Explorer to investigate what's being sent more deeply.

rayshobby, 2021-01-30:
Not sure if this is still an issue as I've not received other reports of this issue.

closed 2021-01-30 by rayshobby