OpenGarage/OpenGarage-Firmware #43: short MQTT server IP

idxman01, 2019-09-03:
All calls checking for the MQTT IP to be >8 characters need to be changed to 6.  While very unlikely, it technically could be 7 as X.X.X.X .  I use subnets under 10.x.x.x and MQTT wasn't enabled until I added a space after the IP, bringing it up to 9 characters.

https://opengarage.io/forums/topic/mqtt-and-home-assistant/

glenford, 2020-04-23:
I hit this problem yesterday and had me confused, it took a while looking at the source to see why it wasn't working.  I have worked around it by adding a leading zero. e.g. 10.0.0.02, this survives a reboot.

glenford, 2020-04-23:
I also hit this issue today.  I have worked around it by adding a leading zeros, e.g. 10.0.0.01
This survived a reboot.

rayshobby, 2021-01-30:
This has been addressed in the upcoming firmware 1.1.2 (still in development branch):
https://opengarage.io/forums/topic/openthings-cloud-otc-beta-test-call-for-help/
it removed the requirement of 8 chars or more.

closed 2021-01-30 by rayshobby