OpenGarage/OpenGarage-Firmware #45: FEATURE REQUEST: Https with Lets Encrypt?

jjanderson, 2020-01-09:
Would be great if there was an HTTPS option, even better if this could be tied into a lets encrypt service so that we can expose the OpenGarage web server confidently... 

Thanks again for a great piece of Kit!

matt199x, 2020-02-11:
I'd recommend just connecting to your network with a VPN. Free options such as PiVPN can quickly be set up. No reason why this should be exposed to the outside world.

rayshobby, 2021-01-30:
Another option is to use the OTC version of the OpenGarage firmware:
https://opengarage.io/forums/topic/openthings-cloud-otc-beta-test-call-for-help/
which is our own cloud service. With OTC firmware, the phone / app connects to the OTC server securely.

closed 2021-01-30 by rayshobby