OpenGarage/OpenGarage-Firmware #46: FEATURE REQUEST: Ability to use reed switch instead of ultrasonic sensor

matt199x, 2020-02-11:
A reed switch is a very accurate way of telling whether or not a door is opened or closed. It will also prevent everyone's garage from being filled with ultrasonic waves which cause some people migraines, especially if you are one to work in the garage a lot.

trevor61, 2020-06-09:
Not certain if this feature was enabled when you requested it but the current version (Firmware v1.1.0 ) supports this type of sensor. See text below from user manual about half way down page 3. Under Options - Door Sensor

> Normally Closed or Open Switch: you can also choose to use external sensor (e.g. magnetic
> switch), connected between pin G04 and GND. Using these options will bypass the built-in
> distance sensor and use external sensor instead.

rayshobby, 2020-08-09:
This is already supported by the current firmware. Closing this issue now.

What will be implemented in the next version is the ability to use both sensors (e.g. reed sensor for door status and ultrasonic sensor for vehicle status etc.).

closed 2020-08-09 by rayshobby