OpenGarage/OpenGarage-Firmware #48: Add capability for MQTT to send closing/opening

basroovers, 2020-04-01:
I use the Home Assistant integration to read the mqtt messages, which works like a charm for me.
But to really top it off, I would need to also receive the closing and opening events.

On the Home Assistant side it would look like this:
```
cover:
  - platform: mqtt
    name: "Garage door"
    device_class: "garage"
    state_topic: "My OpenGarage/OUT/STATE"
    command_topic: "My OpenGarage/IN/STATE"
    payload_open: "open"
    payload_close: "close"
    state_open: "OPEN"
    state_opening: "OPENING"
    state_closed: "CLOSED"
    state_closing: "CLOSING"
    optimistic: false
    retain: false
```

Documentation on this: https://www.home-assistant.io/integrations/cover.mqtt/

idxman01, 2020-04-02:
You’re wanting it to send additional mqtt messages upon status change?  It publishes the status in two topics. 

I also used mqtt in the past, but for now just to use the HA module which relies on the http api.

basroovers, 2020-04-02:
I would like to see these additional mqtt messages between the two states.

State == closed
State == opening (extra mqq message)
State == opened

Vice verse

State == opened
State == closing (extra mqq message)
State == closed

trevor61, 2020-06-09:
The main problem with opening and closing events is you don't know if the event has stalled or been stopped part way or if it has been triggered by entities outside of OpenGarage.  For example by a garage remote or the button on the door motor itself. The intermediate state always has to be assumed (optimistic). The best you can say is the door is closed or the door is not closed, it may be fully open or partially open, depending on how your sensor is positioned. The only way that you can ever be certain the door is moving in a particular direction and the percentage it is open is by using something like a rotary encoder or similar sensor.

rayshobby, 2021-01-30:
The current firmware already sent out notification at the
/OUT/NOTIFY
topic, in addition to the /OUT/STATE topic that reports status every 15 seconds.

closed 2021-01-30 by rayshobby

nicx, 2021-04-20:
@basroovers do you have "solved" this in any way? I am using OPenGarage with the HomeKit integration of Home Assistant, too, and the missing state "opening/closing" makes the cover behave very strange in HomeKit. When clicking on "Open" the cover-button is activated for a very short time and then shows "closed" for the time OpenGarage is opening the door. When this is finished, the HomeKit cover jumps directly over to "Open".

It would be great when open garage "assumes" a state opening/closed depended of the former finale state "open/closed". It should be clear that this is only possible assuming the state, but for HomeKit this is needed to get a right behavior.

@rayshobby this assumed state is not sent to MQTT, do you thing it could be implemented in the firmware?