OpenGarage/OpenGarage-Firmware #49: WiFi setup with special characters or long length

timcary, 2020-04-25:
My SSID is 32 characters long and includes hyphens, but no other special characters.  My password is shorter (12 characters) and does not have any special characters.  I am unable to set it up.  When in AP mode, entering the SSID and password does nothing when I hit submit.... no beep, no refresh. Also, no wifi networks show up.  I added a test SSID to my network that has a short name and password, and it setup just fine. The problem here is that I don't want to have a new SSID just for this device. All my 30 or so other devices are on the other SSID.  Is there any way this can be fixed so the code will accept the long input? I am assuming the issue is the length.  Or, is there any other secret way I can input my long SSID once I am connected with the shorter testing SSID?

psmedley, 2020-04-25:
This might be the issue:
https://forum.arduino.cc/index.php?topic=295728.0

"According to the documentation of the standard, the length of an SSID should be a maximum of 32 characters (32 octets, normally ASCII letters and digits, though the standard itself doesn't exclude values). Some access point/router firmware versions use null-terminated strings and accept only 31 characters."

rayshobby, 2021-01-30:
The firmware does not limit the length of the SSID. This seems a limitation of the ESP8266 chip which doesn't accept SSID of that length. So this isn't something the firmware itself can fix.

closed 2021-01-30 by rayshobby