OpenGarage/OpenGarage-Firmware #52: Feature Request: Ultrasonic sensor & NO / NC Switch

trevor61, 2020-06-09:
Would it be possible to add an option for the ultrasonic sensor to detect if the vehicle is in the garage and simultaneously use a switch to detect if a roller type door is closed or not closed, rather than having them as only mutually exclusive options.

jaseelye, 2020-07-02:
I came here to ask the same thing. I installed a reed switch which allows me to ensure the door is shut. Using the ultrasonic sensor, the door could be partially open but it would report closed. I'd still like to use the ultrasonic sensor to detect vehicle presence. Please add this feature.

tcgerhard, 2020-08-14:
I'd like to see a hybrid -- door is closed if the reed switch is closed, open if the ultrasonic sensor sees the door, and unknown if neither one sees the door.   You could get more specific, and call it "Closing" from when the ultrasonic stops seeing it until the reed closes, and "Opening' in the other direction.

PaperLawyer, 2021-01-02:
I would like this functionality too - I'm currently suffering frustrations that if I trigger opening through Home Assistant the status remains "Opening" (despite putting this code into the custom components folder: https://community.home-assistant.io/t/opengarage-open-source-wifi-garage-opener/17228/175?u=swbradshaw) .  Therefore I'm hoping that use of a reed switch would help.

jduff001, 2021-01-03:
I do not want install a reed switch,  but I have seen, if I initiate a "closing" or "opening" and it does not complete .... then it is stuck in limbo unless restart.  I reports "closing" or "opening"

PaperLawyer, 2021-01-03:
> 
> 
> I do not want install a reed switch, but I have seen, if I initiate a "closing" or "opening" and it does not complete .... then it is stuck in limbo unless restart. I reports "closing" or "opening"

I had the same issue - I'm unsure whether it is the current cold weather but the state within Home Assistant freezes as "Opening" even though I can see the state is either "Opened" or "Closed" within the OG interface.  I also note the OG log is incomplete and the majority of actions are not logged. I can only appear to resolve the "Opening" frozen status by rebooting my Home Assistant device (a thin client dedicated to Home Assistant).

rayshobby, 2021-01-30:
This has been addressed in the upcoming firmware 1.1.2 (still in development branch but will be released soon):
https://github.com/OpenGarage/OpenGarage-Firmware/commits/dev-1.1.2
it supports using both ultrasonic distance sensor as well as switch sensor, and provide an option to choose which sensor(s) you want to use to determine the door status.

closed 2021-01-30 by rayshobby