OpenGarage/OpenGarage-Firmware #55: Feature Request - Add username/password fields for MQTT server

stk21, 2020-07-19:
My MQTT server uses username and passwords, and all of my devices support this.  Can the firmware for this also include support for authentication?

Thanks.

jduff001, 2020-08-13:
I noticed that the mqtt options did not have the username/password options.  Since my mqtt does not allow anonymous I have not been able to test mqtt.  Instead I have been the Home Assistant opengarage cover/template.  This seems OK for one device, but not the second garage door device with a different ip.

rayshobby, 2021-01-30:
This has been addressed in the new firmware (currently in development branch) 1.1.2:
https://github.com/OpenGarage/OpenGarage-Firmware/commits/dev-1.1.2
once we finish testing it will be released. The previous firmwares didn't support this because they used an outdated MQTT library that doesn't support authentication. 1.1.2 has changed the MQTT library to the latest which supports authentication now.

closed 2021-01-30 by rayshobby