OpenGarage/OpenGarage-Firmware #57: Feature Request: Wifi Connect to strongest RSSI rather than first one found.

pandabear41, 2020-08-13:
Been running into problems with the OpenGarage FW where it connects to a weak AP rather that a strong one on same SSID. This is causing issues with delayed Blynk and MQTT. 

I am guessing that `WiFi.begin` without a channel connects to the first channel it detects on the SSID. A potential solution would be doing a wifi scan and finding the strongest RSSI on the specified SSID instead of letting WiFi.begin select it.

jduff001, 2020-08-13:
My local wifi has multiple AP that all broadcast the same SSID (netgear orbit's).  I have enabled "fast roaming" that is supposed to have the router move the device to the strongest AP signal,  but it does not work.  I don't know if the Wifi.begin has the ability to detect the RSSI.
My workaround has been to power on the opengarage when only the nearest AP is operating... that is not really a great workaround.

teufelhunden0311, 2020-08-13:
I know it isn't the greatest solution but I have a multiple AP setup like this with Unifi and to get around this I simply gave the OpenGarage a static IP address and forced it to only connect to the AP I wanted it to. I don't know if this is possible with Netgear products though.

ares, 2020-12-30:
This feature would be great

majorsl, 2024-04-25:
Any chance we'll ever see this?  Even if I use my MESH software to force OG to a particular access point, it doesn't stick.  OG will latch on to an access point and won't give it up without a reset and re-pairing.

Example: I have a WAP in my garage as part of my MESH network.  I updated its firmware and OG went to the next nearest WAP with a very poor signal...and it won't return automatically to the nearest WAP now that it is back on-line.

closed 2026-11-07 by rayshobby