OpenGarage/OpenGarage-Firmware #59: Feature Request: Multiple Device Key Types

smcpeck, 2020-10-02:
Just install OG today and must say that I love it.

It would be great to have an option where I can basically create multiple device keys with different access levels.

Admin Key:  Allows to change settings on OG and perform basic operation.
User Key:  Only allows open/close of the door.

Currently, if I give the device key to a guest/tenant, they have the ability to really go in and goof things up.  It would be nice to avoid that.

If I'm missing something, I'd be delighted.  Also happy to help out and throw up a PR if this would be a quick change on the code side -- I'm somewhat capable even thought I have only toyed lightly with an Arduino and haven't touched C++ in 15-20 years. 😆 

Cheers!

idxman01, 2020-10-02:
I would have them use the Blynk app or integrate it into another platform like Home Assistant.  From there they won’t need to have the key.

smcpeck, 2020-10-02:
Thanks, @idxman01, I'll goof around with that a bit.  I enjoyed avoiding Blynk altogether, but can see how that might help my situation.

I saw that there's a new alpha app out that uses the OTC tokens, too... maybe that would fit the bill once it is fully ready.  🤔 

Thanks again for the suggestion! 👍

smcpeck, 2020-11-17:
FYI - Blynk app has been fitting the bill here for now. I'd still like to see a more native solution to this use case as the Blynk app is a little clunky/intimidating for non-technical folks.

closed 2020-11-17 by smcpeck