OpenGarage/OpenGarage-Firmware #62: Any firmware past 1.0.8 will not connect with Blynk

teufelhunden0311, 2021-02-09:
How do I get past this? I have left my OpenGarage on 1.0.8 for so long because anytime I upgrade to a newer firmware I keep getting Generic Board offline errors in Blynk and even getting a new key and everything does nothing to fix it while going to the GUI via the IP address on my computer works fine.

rayshobby, 2021-02-09:
I cannot reproduce this issue: there are at least tons of users out there using firmware 1.1.1 and I have not received any report of their firmware not connecting to Blynk.

teufelhunden0311, 2021-02-09:
[image]

teufelhunden0311, 2021-02-09:
That is the error that pops up for any firmware past 1.0.8. If I go back to that firmware it will just start working again

rayshobby, 2021-02-09:
Are you aware that updating firmware may trigger a factory reset, hence the controller will be reset back to factory reset? So you will have to go through the WiFi setup steps to get the controller onto your WiFi router again.

teufelhunden0311, 2021-02-09:
I mentioned in my post how the GUI is working like normal, same password, same static IP address and everything.

rayshobby, 2021-02-09:
what's the version of your blynk app

teufelhunden0311, 2021-02-09:
2.27.24, most up to date according to the google play store

rayshobby, 2021-02-09:
i have exactly the same blynk app version, and it works fine with my firmware 1.1.1. so I honestly have no idea...

teufelhunden0311, 2021-02-09:
quite unfortunate. I just don't know why the version matters. If it stopped working on the previous version I would know it was a network issue on my end but something has to change in the firmware that then causes it to fail to connect. But o well back to 1.0.8

rayshobby, 2021-02-09:
actually: one thing you can do is to check the information the firmware is sending to the blynk cloud server, by using the blynk http api:
https://blynkapi.docs.apiary.io/#
it needs you to plug in the blynk token to the url. For example you can use the get pin api to see the value the firmware is reporting to the server. Pin numbers are defined in the firmware:
https://github.com/OpenGarage/OpenGarage-Firmware/blob/master/OpenGarage/defines.h#L85
V0 is door status, V3 is distance value etc.

There is also a get project api which can return the information of the entire project.

If the firmware is not reporting any data to the server, you may want to reboot the controller and let it reconnect.

teufelhunden0311, 2021-02-09:
I'll check it out. Maybe if a new firmware ever comes out that will just fix it somehow. It works fine for me on 1.0.8 though so I'm not too worried about it. I figured I would just ask the developer this time to see if they had any insight to this issue.

idxman01, 2021-02-09:
Ray just merged a ton of updates so I suspect the next firmware will be coming soon.

As for Blynk, I've been using 1.1.1 for a year and a half and it's fine. However, I do see it report a disconnect from time to time, but it'll connect again pretty soon.

closed 2021-02-09 by teufelhunden0311