OpenGarage/OpenGarage-Firmware #63: FW 1.1.2 Beta - MQTT non default port bug

SamForma, 2021-02-16:
Running Firmware 1.1.2

When trying to connect to a mosquitto broker listening on a port other than 1883, the connection is not made. I see nothing logged in the mosquitto log nor are any messages received by subscribers. 

to rule out firewall related issues, I issued a message to a new topic using mosquitto_pub while the mosquitto service is running on 1884, and was received fine by mosquitto_sub. Connections are logged properly.

MinimalPTH, 2021-09-26:
Thank you for posting this report - I just spent several hours trying to understand why I was not able to publish to my MQTT server on a non-standard port.  Here's hoping for a quick resolution.

jduff001, 2021-09-26:
Please clarify if port 1883 works.. but a custom port is not working?

MinimalPTH, 2021-09-26:
Port 1883 works.  Custom port does not work.

Normanras, 2021-11-03:
Hi - I recently bought OpenGarage and just added a DHT22 to the sensor to G05 and confirmed that it is working via the web GUI. But I am not getting any information via MQTT. Either via Home Assistant or the MQTTool app. I can confirm that my other MQTT events are firing and being picked up, but OpenGarage is not transmitting anything. Server, port, user and pass are all correct and working with other devices.

rayshobby, 2022-10-24:
This issue should be fixed in the now released firmware 1.2.0. Previously 1.1.2 beta was using fixed port 1883. This has now been fixed.

closed 2022-10-24 by rayshobby

sumoneelse, 2022-10-25:
Thank you!  Unfortunately I cannot use 1.2.0 as I'm on an older hardware version. I'm glad others will benefit from this fix though.