OpenGarage/OpenGarage-Firmware #75: Socket Hangups

pixeleyesltd, 2022-10-19:
Since moving to the latest 1.2.0 firmware I'm setting continual socket hang ups via the OpenGarage Homebridge plugin.

```
[10/19/2022, 2:38:00 PM] [Garage] Error getting state: Error: socket hang up
[10/19/2022, 2:38:00 PM] [Garage] Error getting state: RequestError: Error: socket hang up
    at new RequestError (/homebridge/node_modules/homebridge-og/node_modules/request-promise-core/lib/errors.js:14:15)
    at Request.plumbing.callback (/homebridge/node_modules/homebridge-og/node_modules/request-promise-core/lib/plumbing.js:87:29)
    at Request.RP$callback [as _callback] (/homebridge/node_modules/homebridge-og/node_modules/request-promise-core/lib/plumbing.js:46:31)
    at self.callback (/homebridge/node_modules/homebridge-og/node_modules/request/request.js:185:22)
    at Request.emit (node:events:527:28)
    at Request.onRequestError (/homebridge/node_modules/homebridge-og/node_modules/request/request.js:877:8)
    at ClientRequest.emit (node:events:527:28)
    at Socket.socketOnEnd (node:_http_client:478:9)
    at Socket.emit (node:events:539:35)
    at endReadableNT (node:internal/streams/readable:1345:12) {
  cause: [Error],
  error: [Error],
  options: [Object],
  response: undefined
}
```

rayshobby, 2022-10-23:
Sorry, I have no knowledge about the Homebridge plugin. Given that the main change of 1.2.0 is adding support for OTC, if you don't need to use OTC, you can go back to firmware 1.1.2.

psmedley, 2022-10-23:
Hi Ray,

It's interesting, but this sounds similar to what I'm seeing when OTC is 
used in conjuction with openhab. In my case with openhab, it's simply 
calling http://opengarage/jc every 60 seconds...

Cheers,

Paul

rayshobby, 2022-10-23:
I don't use OpenHab, but I did the test by using a script I wrote myself and request http://opengarage/jc every 60 seconds. I didn't observe any persistent problem -- over a day there maybe a few call failures but it doesn't get stuck, it recovers and the request continue to succeed.

rayshobby, 2022-11-23:
I just posted this question in the other issue thread:
I am wondering if the homebridge plugin is sending a request that is not GET request (i.e. it might be a POST or PUT request etc.) The reason I am asking is that after some debugging, I found the OTF library has a bug that results in any non-GET request returning a 413 error. The normal use case of OpenGarage shouldn't trigger this because all requests are indeed GET requests. But since you have a plugin that's sending request to it, I am not sure what it's sending to the controller, and it can certainly trigger this bug.

Derpthemeus, 2022-11-25:
Error is caused by a timeout when making a GET request to /jc

https://github.com/OpenGarage/homebridge-og/blob/ae072c4958555f97c4f75102cc8b8b963270f935/lib/open_garage_api.js#L19

Derpthemeus, 2022-11-25:
@pixeleyesltd are you able to provide serial logs from the OpenGarage device?

closed 2023-01-05 by rayshobby