OpenGarage/OpenGarage-Firmware #83: Documentation: Firmware 1.2.0 Get Options API is missing a few variables

jknaack, 2023-07-02:
The Firmware 1.2.0 GetOptions API (section 4 in the doc) is missing 2 variables in the description.
Documentation source: https://github.com/OpenGarage/OpenGarage-Firmware/blob/master/docs/fw1.2.0/OGManual1.2.0.pdf

1) dns1 - This is the DNS server to use when using the Static IP mode.  It's even available to edit in the editor, but not documented.   It should probably be listed with dvip/gwip/subn.
https://github.com/OpenGarage/OpenGarage-Firmware/blob/8fab4c65a1b2b9b8b57659c1ddd3ba0f9b26816b/OpenGarage/OpenGarage.cpp#L85

2) mod - This is probably not really important, since it's the currently running WIFI mode - either AP or STA.  In practice, if you are using this API, you are probably in STA (I haven't tested if it's possible in AP).  Still, for completeness, it would be good to document
https://github.com/OpenGarage/OpenGarage-Firmware/blob/8fab4c65a1b2b9b8b57659c1ddd3ba0f9b26816b/OpenGarage/OpenGarage.cpp#L60

I would be happy to make these changes and submit a merge request, but the documentation is only published in PDF format, and I don't have access to the original source.  If the source is available somewhere, please let me know and I can edit it there and publish a new version.  Alternatively, if you are interested, I could take the PDF content and move it to a OGAPI1.2.0.md Markdown file, and publish that instead (or beside the pdf?) so it's easier for others to submit changes.

rayshobby, 2023-08-24:
Thanks for the suggestions. Your first link is to the user manual, I assume you meant the API document PDF instead. In any case, I updated the API document and added `dns1` option. I didn't add the `mod` option because it's not modifiable by the user via the `co` command (in fact, it shouldn't be shown on the `jo` page either).

closed 2023-08-24 by rayshobby

jknaack, 2023-08-25:
I don't know how I got the wrong link, but yes, I meant the API docs.