OpenGarage/OpenGarage-Firmware #85: Feature Request: Add support for DHCP

jknaack, 2023-07-30:
The current implementation supports mDNS, but not DHCP, which frequently confuses routers, older windows PCs, etc.  It would be nice to support DHCP as well as mDNS when connected to a network.

rayshobby, 2023-08-24:
The current firmware supports DHCP, in fact that's the default configuration unless if you turn off DHCP and use static IP. I looked at the PR you linked to, it seems what you meant is a different way for DNS in addition to mDNS.

closed 2023-08-24 by rayshobby

jknaack, 2023-08-26:
Wow, my title was terrible, sorry about that.  I should have clarified that I was setting the DHCP hostname, not just using DHCP.  Would you be interested in taking another look at this feature request with that in mind?

Currently, on several router I've tested, I noticed that the name on the router is different than the network DNS name.  I've seen this on an Asus router, an OpenWrt Router, a Linksys, and a Unifi router, and they all show the name as ESP-ABC123 (6 digits of the mac) regardless of what the mDNS hostname is.  I also ran a tcpdump and confirmed that the DHCP hostname registration is the default ESP-ABC123 name even if the hostname is set.  If you have multiple devices, this makes it difficult to differentiate which ones you are looking at in your router settings, and is also just generally confusing.  

If you look at the [example docs for mDNS](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266mDNS/examples), you can see that most of them also set the DHCP hostname, likely to avoid this problem.  I don't know what other problems, if any, may happen with having a different name registered on the DHCP server than in the DNS , but at a minimum, it's confusing.  For example, in the Arduino mDNS-SD_Extended code, they set the hostname [here](https://github.com/esp8266/Arduino/blob/1a4663fbe8dca5ee3992aa7b0fc98794d956ac30/libraries/ESP8266mDNS/examples/mDNS-SD_Extended/mDNS-SD_Extended.ino#L32) and begin the mDNS server with the same host name a [few lines down](https://github.com/esp8266/Arduino/blob/1a4663fbe8dca5ee3992aa7b0fc98794d956ac30/libraries/ESP8266mDNS/examples/mDNS-SD_Extended/mDNS-SD_Extended.ino#L46).  This same pattern is repeated in several other examples.  There's also some commentary in the [Arduino discussion boards ](https://github.com/esp8266/Arduino/discussions/8816#discussioncomment-4767062)about how the wifi hostname and mdns hostname are not the same, which is the issue I was trying to address.

Note: Setting the name on the AP page was really just a convenience to avoid having to wait for the DHCP name to time out on the router so you see the name change immediately, but otherwise didn't have anything to do with the change.

PR #84 (2024-12-16): Adding support for DHCP name resolution so that routers and other com…
…puters can connect without requiring mDNS.  This is especially helpful for routers that record the DHCP name in their ip table, now they will show the right name.  Also adding the hostname as an option in the ap_home page so you can specify the dhcp host name as soon as you connect to your network

This should fix #85