OpenGarage/OpenGarage-Firmware #89: Feature Request: Allow Switch Sensor to be used for "fully closed" door status

mgolik, 2023-08-16:
Switch sensor can currently be used to determined "closed" status (with, without and in conjunction with distance sensor).

What I would like to achieve is use distance sensor for "fully open" state (and car present / not present) and switch sensor for "fully closed". With that we could have "fully open", "fully closed", "open (not closed)" states and also "opening" and "closing" when triggered from app.

This was also mentioned in this topic - https://github.com/OpenGarage/OpenGarage-Firmware/issues/52 , specifically https://github.com/OpenGarage/OpenGarage-Firmware/issues/52#issuecomment-673837767

jknaack, 2023-08-23:
I've been slowly working on an implementation for something so that you can say "I'd like to open my garage door 50% open".  I have described this as "partially open".  Is that the same idea as to what you are describing?  I currently have a version where you can specify a fixed time to open your door (ex: open for 3000ms), which is a good start, but I'm hoping to get back to it eventually.

Note: I'm not a maintainer.

mgolik, 2023-08-25:
This also sound like a nice feature but what I am interested in this ticket is a way to make sure that doors are 100% closed - because currently I know when they are 100% open since distance sensor is at the end of the gate. But when I open the door by hand for example for 50% I will have status "closed" yet the doors will be open.
So my request along with your feature would work together nicely but by itself it does not resolve my issue.

jknaack, 2023-08-25:
If you set the "Switch Sensor" setting to "Normally Closed" and the "Sensor Logic" to "Dist. or Switch" you would get a status of "open" if the garage door is partly open, and would only say closed if it is fully closed (but you would lose Fully Open, which sounds important to you as well).  That's the closest solution I've come up with so far.

BillyCroan, 2023-10-31:
I second this.  I'd like to put a limit switch in the track so when the door is fully down, the lowest wheel just hits it.  It should fail if for example, snow falls into the door and it's an inch high.

Today I use a live video feed to monitor the door when I command it to move.  But if the OpenGarage Hardware can receive a signal from a reed switch, let's put it to use.  And here I was daydreaming about putting a QR code on the door and using a program to check its position in the camera feed....

As an alternative, I might suggest using two OpenGarages.  one at the beginning of travel and one at the end.  So you can determine intermediate state if the two don't line up.  You could also wire both in parallel and use different wifi networks to add some redundancy.

rayshobby, 2024-10-10:
If I understand it correctly, what's being requested sounds more or less like an UI feature: because the UI already reports both the distance sensor status and the switch sensor status, it's just a matter of labeling different combinations of them, something like: if distance sensor says door is closed, switch sensor says it's open, then it's 'partly open'. if both says closed, then it's 'fully closed'. The exception is: if distance sensor says open, switch sensor says closed, that seems a state of discrepancy. In any case, it sounds like a UI-only change.

PR #99 (2024-10-15): Email support
resolve #90 
resolve #89 

This version of firmware adds Email notification support. also addresses issue 90 regarding time keeping functions.
closed 2024-10-15 by rayshobby