OpenGarage/OpenGarage-Firmware #93: Empty log

BillyCroan, 2023-10-31:
My OG's log page shows no events.   I restarted the OG, the operated the door from the wall button and again by the app.  But at no point did it log any events.  I checked and the time displayed on the log page is correct and the ntp server configured is working properly.  It's as if logging on the OG is just turned off.  ( I found https://github.com/OpenGarage/OpenGarage-Firmware/issues/90 while checkin to see if others reported logging problems )  NTP is configured and working fine for me though.

Would it be possible to get the OG to log via syslog to a remote syslog server?  it's a single udp packet per event I believe, and this would relieve the OG of having to store log data.  in fact, syslog servers can often put their own timestamps on events so the OG wouldn't even need to have an accurate clock then.

In the meantime, I set log length to 20 then cleared log and reboted
then set the log to 400 and then cleared log and rebooted again.  Still no events.

LSWON, 2023-11-02:
I am having the same issue. No logs displayed.

BillyCroan, 2023-11-02:
@LSWON Does the web interface of your OG show clearly that your door is open when its physically open, and that its closed when its physically closed?

What are your thresholds set to?  Mine were: 50 and 150
[image]
And that was causing my problem.  I could control the door, but what I hadn't noticed was the button on the interface always said "Open Door" because it was always thinking the door was closed.  

The only events it ever logs I think, are door open and close events.  It would be nice it it logged when the door was _commanded to operate_ via the OG device or manually (if that can be sensed), but it doesn't log that.  It only logs what it can interpret from the thresholds you have set.

I changed my thresholds to 80 and 200, and now it works.  But what's correct for me and you will be different because our garages are different.  Refer to the manual on how to choose thresholds.

Check what distance it reads with the door closed and car gone (MAX).  Then what it reads with door closed and car in garage (A), lastly, what it reads when the door is open (B).  Your door threshold (in my opinion) should be the average of A and B.  And car threshold the average of MAX and A.  

It would be nice i this suggestion appeared on the administration screen.  I wouldn't even tell the user the unit is centimeters, as that would lead people to measure with a tape, as I did.  Your sensor could be out of calibration, dirty, affected by humidity or pressure.  So it might not match with actual measurement by hand.  Better to use the values you see in the OG webpage for your measurements.

LSWON, 2023-11-02:
Thank you so much for replying. I changed the numbers around and it's working as intended now.

BillyCroan, 2023-11-02:
Glad to hear it. I'm a little embarrassed that I was chasing the wrong problem but I'm going to leave this up to help others in the future.

closed 2023-11-02 by BillyCroan