OpenGarage/OpenGarage-Firmware #94: OpenGarage/HA gets confused by partial door open and then close

TopBanana123, 2023-11-01:
I'm wondering if anyone has run into this, I am not sure if its an HA issue or an OpenGarage issue

If my garage door is closed, and I hit the button to open it, then press again to stop it, and then press again to close it, it gets stuck in an “Opening” state in HA

I do this from time to time, sometimes because I accidently hit the garage button to open but more often, to scare away Raccoons from the cat door

Right now OpenGarage shows CLOSED, which is correct

[image]

However, HomeKit shows Opening with the spinning circle

[image]

And in HA, it never shows the “Closed” message, which I guess is why HomeKit is showing it in progress

[image]

Is there a way to get it to poll OpenGarage every X seconds to figure out the real status?

TopBanana123, 2023-11-02:
This is what I did to fix it, if anyone has the same problem. It would be handy if this was a built in fix

[image]

rayshobby, 2024-01-02:
It sounds like this is an HA issue, as the OG firmware reports the correct state but HA doesn't. So I am closing this issue for now.

closed 2024-01-02 by rayshobby