OpenGarage/OpenGarage-Firmware #97: Network Timeouts

EPICmynamesBG, 2024-05-30:
I've had OG running successfully for about 2 years now, but within the last week, it's become unresponsive. When I navigate to the IP address, network calls hang until eventually timing out. On the rare occasion, I will get a response after minutes of waiting. I've spent hours debugging now and am at a loss. I will outline my process below; I'm hoping someone else has ideas on what the problem could be.

Background & Environment
- Unifi Network. OG is allocated a static IP via DHCP on my router. OG connects to a dedicated IOT wifi with which resides on it's own VLAN. VLAN _does_ have internet access. 
- Home Assistant - I usually operate OG through Home Assistant. For purposes of debugging, I've shut off Home Assistant and taken it out of the equation.
- I believe I have OG hardware version 2; using the `/db` endpoint, I _eventually_ got it to resolve once after many hours of debugging, and learned It's running firmware version 112.

OG has worked happily for ~2 years in the above setup.

Begin debugging....
Verify network connectivity
- I run a Unifi UDM-Pro at home. Via the console, I could see the device was powered on and connected via the static IP I had allocated it within my router.

Reboot/Power Cycle
- Device reconnects to the WiFi just fine. Still gets network timeouts when trying to go to the web page at it's IP.

Attempt to ReFlash via USB
- No luck. Steps: Plug in micro USB to OG, hold button on OG, plug USB into laptop. Result: Little blue light flashes once, quickly. No further activity after this. Computer is unable to detect any serial device connected. Tried via both Ubuntu and Windows 11.
- tried various Micro-USB to USB cables, all same result.

Hard reset - held the button for 10 seconds.
- Progress? Device made noise and went into AP mode. Connected and input Wifi credentials; first time, the device appeared to lock up and hang on submit. Power cycled, tried again, and it worked; device is now connected to the IOT wifi again, getting the same static IP.
- Attempting to configure the device, I'm back in network timeout land. I got to the firmware upload screen a 2 or 3 times, but each time on uploading 1.1.3 FW, the device hung for 10 minutes before I power cycled. I was able to get to the /db page twice.

So after factory resetting, I'm able to connect ~5 times out of 100s of attempts over the course of 2 hours..

Is there anything else I can do here?

EPICmynamesBG, 2024-05-30:
I've wondered if this is related to #90 , but my network conditions are a bit different and OG _does_ have internet access, so I don't believe this is a duplicate.

russellmilliner, 2024-05-30:
I am just another user, but I have a similar setup.  Unify/OG/HA.  My OG is on 1.2.0FW though.  I did have issue with lots of IoT devices on some versions of Unify firmwares.  Just curious, what unifi wifi device(s) are you using and what firmware are they running?
Maybe try running a hotspot on a cell phone and connect the OG to it and see if the behavior is different.

EPICmynamesBG, 2024-05-30:
> I am just another user, but I have a similar setup. Unify/OG/HA. My OG is on 1.2.0FW though. I did have issue with lots of IoT devices on some versions of Unify firmwares. Just curious, what unifi [...]

:wave: @russellmilliner I have 1 U6-LR and 2 U6-Lite APs. Specifically, the OG is connecting to the one U6-Lite AP I have ceiling mounted in the garage ~5-10ft away from the OG. It was funny, during the connection process, I did see my IOT network twice with 2 different signal strengths; I chose the stronger. 

Right now, Unifi reads OG as connected with signal strength -52dBm on 2.4GHz band, 20MHz channel width. 

UDM-Pro running UnifiOS v3.2.12, Network app version 8.1.127.
All APs running FW 6.6.65. 

I do have dedicated "nicer" settings on the IOT Wifi network to make it more compatible with many of the low power, older wifi chipsets used. ie: disable Band Steering, disable Hide WifiName, disable Fast Roaming, Security Protocol WPA2...

I will try what you've suggested to see if it makes a difference. :+1:

russellmilliner, 2024-06-04:
Any update on whether it made a difference?  Also there was a new Network app version recently released.  Mine is connected to a U6 AP Ch. 1 (2.4 GHz, 20 MHz)-46 dBm with same AP FW, so we're in a very similar setup.  My AP is in the room right above the garage.

Something else you can try...I had a bunch of issues with one of the Unifi AP 6.5 FWs connecting Shelly IoT devices and downgraded to a previous 6.4 for a while.  You could always try something like that on the specific AP that your OG is connecting to.

Did you have any luck with the phone hotspot?

EPICmynamesBG, 2024-06-05:
Thanks for the advice, I’ll see if my Network app has updated. Hoping to be able to try it this weekend. Last weekend was busy and work has been all consuming this week. I’ll keep this thread up to date when I do test it 👍

EPICmynamesBG, 2024-06-07:
Following up:

Unifi Network App has updated to v8.2.93. Whole network had to be rebooted today due to an ISP outage combined with the USP-Plug on my  modem failing. As a result of the above, I checked in on the OG at the static IP it is given and it's quickly and happily responding now.

My previous post about firmware was incorrect. The actual version was 1.1.2. I'm playing around with it tonight trying to update firmware and ensure it's stability, but 🤞 it was just a unifi issue?

russellmilliner, 2024-06-07:
very possible it was just a unifi issue.  Sometimes when I have some weird network issues I restart unifi or just the APs.  Just another diag step in the box :-)

EPICmynamesBG, 2024-06-07:
Marking this as closed as it seems like it was a Unifi issue. Odd that OG was the only device on my network impacted, but I'm happy to learn my OG isn't bricked. Thanks all!

closed 2024-06-07 by EPICmynamesBG