OpenGarage/OpenGarage-Firmware #98: Device constantly making DNS requests for NTP servers

rjfendricks, 2024-06-26:
I think this issue is related to, but distinct from, issue #90 . I've noticed that my OpenGarage device is making DNS requests to various NTP servers every couple of seconds, leading to something like 800 DNS requests per hour. This makes it easily one of the most noisy DNS requesting devices on my network for essentially doing nothing but updating its clock. I saw in the other issue that the device doesn't have a real time clock, which is the reason that the NTP servers are queried so often. I recognize the importance of having logs with accurate times, but I do wonder if the clock drift is severe enough to warrant such frequent updates.

[image]

I took a glance at the code, and I think it may have something to do with the `time_keeping()` function found [here](https://github.com/OpenGarage/OpenGarage-Firmware/blob/9d9cc902b2fac4b882d133de18725cc1910c842a/OpenGarage/main.cpp#L1254). It's not entirely clear to what about this code is making the queries happen so regularly. My first guess is that the call to `time(nullptr)` [here](https://github.com/OpenGarage/OpenGarage-Firmware/blob/9d9cc902b2fac4b882d133de18725cc1910c842a/OpenGarage/main.cpp#L1277) is accessing the NTP server under the hood every time it is called, causing a DNS request to be performed.

As mentioned before, my first instinct for solving this would be to limit the number of queries being made to the NTP servers. An alternative solution would be to somehow cache the IP address of the NTP server and use that after the initial DNS resolution is made. Turning off NTP altogether, as suggested in the other issue, would also technically work, though that does feel like a bit of an extreme solution.

rayshobby, 2024-06-26:
It seems to me it's still fundamentally because your device isn't able to get a valid NTP response: In this loop:
https://github.com/OpenGarage/OpenGarage-Firmware/blob/9d9cc902b2fac4b882d133de18725cc1910c842a/OpenGarage/main.cpp#L1279
it calls time(nullptr) every 2 seconds for up to 30 seconds, until it gets a valid NTP time value. If your device has Internet connection and nothing is blocking traffic to it, it should get a valid response on the first try. If so the next ntp call would be in TIME_SYNC_TIMEOUT seconds, which is by default half an hour. 

Are you blocking Internet or having a firewall that blocks incoming traffic to OpenGarage somehow? Obviously when the firmware was written, I didn't expect it would be a common case where it can't get valid NTP time. You can always modify the firmware to bypass the time_keeping function, or reduce the frequency. The firmware is written for the more common cases where the device is connected to the Internet and can get NTP result successfully.

rjfendricks, 2024-06-26:
Ah, your comment led me to find the root cause of the issue I'm seeing. By all accounts it looked like the NTP Domain name was being resolved properly: PiHole saw the requests come through and resolved it successfully and I was able to ping the domain name from another computer on the same network as the OpenGarage.

However, I realized that I'm forcefully sending all DNS traffic (port 53) on my network to my PiHole, regardless of whatever the device sets its DNS Server to. To see if this was causing an issue, I changed OpenGarage's DNS Server from 8.8.8.8 to the IP address of my PiHole, and that seems to have resolved the issue!

I've encountered a handful of devices in the past that really don't like it when it receives a DNS response from an IP address that it wasn't expecting, and I guess this is one of those cases. It should work in theory, but I have no idea how to go about actually resolving this particular issue. Honestly, this one might be so esoteric that it's not worth trying to resolve, maybe just make a note about it for anyone else who comes along that is also forcefully redirecting their DNS traffic.

closed 2024-10-10 by rayshobby