OpenGarage/OpenGarage-Firmware #100: Blynk connectivity issue

joaquinagomes, 2024-10-22:
Blynk persistently disconnecting after some time.
Apparently issue #90 was not fixed in the latest compiled firmware. Try rebooting your router and check if OG hangs and only works again after powering ESP off and then powering on.

rayshobby, 2024-10-22:
Just to be clear: is this firmware 1.2.3 you are referring to? I am very confused by your description. Issue #90 is regarding using OpenGarage without Internet connection, whereas you are reporting an issue with Blynk, which requires Internet connection. Therefore I don't know why this is linked to issue #90, unless if you are using Blynk without Internet? Could you elaborate?

joaquinagomes, 2024-10-22:
> Just to be clear: is this firmware 1.2.3 you are referring to? I am very confused by your description. Issue #90 is regarding using OpenGarage without Internet connection, whereas you are reporting [...]

Exactly, the latest firmware 1.2.3. Maybe I forgot to do a factory reset and configure it again, it even might be a problem with my D1 pro mini. I will check if it's not hardware related, but the issue I'm having is Blynk connectivity. It keeps disconnecting from cloud after about one full day and once I reboot the device it connects normally to Blynk but disconnect again after about one day. The other issue is related to internet or wifi connectivity. If device is deprived from internet connection or wifi even for a few minutes for example after router reboot, OG hangs and only works again after physically powering off and on. There was no Blynk connectivity issue with the firmware 1.2.1.

rayshobby, 2024-10-22:
Well, losing Internet is not the same as losing the router connection -- if Internet is lost, the device is still connected to the router, it still has a valid IP, it can be pinged on the network, it just doesn't have access to the Internet; if the connection to the router is lost, the device will lose its IP address and it won't respond to pings. Your description says that in either of these two cases, you would have to reboot OpenGarage. Is that what you mean? Issue #90 is specifically about losing Internet connection, but the router is still alive and the device should be pingable within the network. Firmware 1.2.3 specifically addresses this issue and I've tested it multiple times.

If the router rebooted or for whatever reason the connection to the router is lost, then after the WiFi connection recovers, you need to reboot to get Blynk working again, that's a separate issue from the above. I'm not entirely sure as I've not specifically tested this.

But the bigger question is why it disconnects in the first place. If it lose Internet connection, the question is why the router loses Internet connection to begin with? If the router reboots, the question is why it reboots in the first place? Maybe try a UPS to see if it helps? 

Also, sadly we will likely discontinue the support for Blynk soon. There are multiple reasons: all iOS users who don't already have Blynk legacy app installed can not install Blynk legacy anymore, so most iOS users are not using Blynk; the Blynk in-app push notification has stopped working, I am not sure why but this has been reported by at least a few users; the main advantage of Blynk is the in-app push notification, since firmware 1.2.3 added email notification, which can replace push notification, there isn't a strong motivation to keep supporting Blynk (on top of that, we have to maintain our custom Blynk server). For all these reasons, we will most likely remove the support for Blynk at some point, and only support OTC from that point on.

joaquinagomes, 2024-10-22:
> Well, losing Internet is not the same as losing the router connection -- if Internet is lost, the device is still connected to the router, it still has a valid IP, it can be pinged on the network, [...]

I remember with previous firmware versions, there was no problem at all with or without internet connection even if router was turned off and there was no wifi available for a while, once the wifi SSID turned up, the OG would connect automatically. I mean with previous version OG was persistently trying to connect to the wifi SSID assigned to it. Now it only requires a router reboot and OG won't connect to the wifi anymore until I physically reboot ESP device.

rayshobby, 2024-10-23:
The only change I can think of, that relates to what you described is this line:
https://github.com/OpenGarage/OpenGarage-Firmware/commit/ee6709e3b04bc4f46864a2f93838b1a60325ff1d#diff-1d28f6135c4b4920ea0e9be695af8d7d75ff11d7e2f36b96d59bd4347ad3995bR1515
`Blynk.connected()` was added to the condition to fix a minor issue that causes problems when Blynk token has just been configured but the controller hasn't rebooted yet. It's possible that if `Blynk.connected()` fails (in the case the router or internet connection is lost), it will no longer call `Blynk.run()` which perhaps manages the reconnection. However, this doesn't explain why the controller doesn't remain on the WiFi anymore, since WiFi is not managed by Blynk itself. In any case, if you can compile the firmware yourself, you can remove the `Blynk.connected()` condition and see if it makes any difference.

joaquinagomes, 2024-10-23:
> The only change I can think of, that relates to what you described is this line: [ee6709e#diff-1d28f6135c4b4920ea0e9be695af8d7d75ff11d7e2f36b96d59bd4347ad3995bR1515](https://github.com/OpenGarage/OpenGarage-Firmware/commit/ee6709e3b04bc4f46864a2f93838b1a60325ff1d#diff-1d28f6135c4b4920ea0e9be695af8d7d75ff11d7e2f36b96d59bd4347ad3995bR1515) `Blynk.connected()` was added to the condition to fix a minor issue that causes problems when Blynk token has just been configured but the controller hasn't rebooted yet. It's possible that if `Blynk.connected()` fails (in the case the router or internet connection is lost), it will no longer call `Blynk.run()` which perhaps manages the reconnection. However, this doesn't explain why the controller doesn't remain on the WiFi anymore, since WiFi is not managed by Blynk itself. In any case, if you can compile the firmware yourself, you can remove the `Blynk.connected()` condition and see if it makes any difference.

I will try that out. I will also test with another ESP hardware to try to figure out the wifi connectivity issue. Thank you very much. Congratulations for the support and high quality open source project.

joaquinagomes, 2024-12-19:
> The only change I can think of, that relates to what you described is this line: [...]

This solution worked 100%, after removing 'Blynk.connected()' and compiling the firmware myself and several days of testing, there is no Blynk connectivity issue anymore.

rayshobby, 2024-12-20:
OK, thanks for letting me know. I will change it back in the next firmware release.

PR #108 (2025-10-07): Firmware 1.2.4
close #106 
close #101 
close #100 
close #71

This firmware adds support for a new hardware version OpenGarage 2.3+, which can natively talk to Security+ 2.0/1.0 without an external adapter.

- Auto-detection of Security+ 2.0/1.0.
- Reporting a complete list of door states: closed, open, stopped (partially open), closing, opening.
- Control opener's light and remote lock (enable/disable garage remotes).
- Obstruction sensor reporting.
- Wall-panel emulation for Security+ 1.0.
- Redesigned AP-mode WiFi setup page with a sleek new appearance.
- Built-in web UI now serves minified [... cut]
closed 2025-10-07 by rayshobby