OpenGarage/OpenGarage-Firmware #101: Status change delay

ads1230, 2024-11-23:
Hi, I've set up my OpenGarage using a contact sensor to detect open/close

My old setup involved an Aqara door/window sensor to detect the door status and a spare garage remote hooked up to an ESPP board.

The status changes on my old setup were pretty much instant, as soon as it closed or opened the status would change.

Using opengarage the state change definitely has ~ a 10s lag before updating.

Is there a way to make the device poll quicker? I would understand a late with the distance sensor but not a wired open/close.

rayshobby, 2024-11-23:
This can be improved. The lag is technically only needed if you are using the built-in distance sensor. Because the distance sensor can be noisy, and transient events like a person walking below it can trigger a false positive, it uses multiple reads separated by 'read interval' between every two sensor readings, to filter out noise. The switch sensor reading is handled in the same function as distance sensor, as a result, it also incurs this long lag, which I agree is not necessary. This can certainly be modified in the firmware, by decoupling the reading of switch sensor from reading of distance sensor. In the meantime, you can reduce the 'read interval' value, which is in Advanced tab of options, and that should reduce the lag.

ads1230, 2024-11-23:
That would be fantastic if the setting could be decoupled!
I will try adjust the read interval to reduce the lag.
I'm sorry I'm not much help on the coding side.

Also, out of interest, I wired my Aqara sensor up to the board as the open/close sensor. When disconnected from the board it updates via zigbee on state change, when hooked up to the board it remains on closed. Could this be due to the voltage the board outputs through the sensor affecting the state change?

PR #108 (2025-10-07): Firmware 1.2.4
close #106 
close #101 
close #100 
close #71

This firmware adds support for a new hardware version OpenGarage 2.3+, which can natively talk to Security+ 2.0/1.0 without an external adapter.

- Auto-detection of Security+ 2.0/1.0.
- Reporting a complete list of door states: closed, open, stopped (partially open), closing, opening.
- Control opener's light and remote lock (enable/disable garage remotes).
- Obstruction sensor reporting.
- Wall-panel emulation for Security+ 1.0.
- Redesigned AP-mode WiFi setup page with a sleek new appearance.
- Built-in web UI now serves minified [... cut]