OpenGarage/OpenGarage-Firmware #102: Inoperable with no internet

pixeleyesltd, 2025-01-13:
We had an ISP outage for 24h, and during that time the OpenGarage web UI was inoperable. Additionally, the API returned that the door was open, even though it was not. 

This is reproducible with firmware v1.2.3. The only integrations configured are MQTT and HTTP API, cloud connectivity is disabled. Screenshot attached to show the issue with the web UI.

[image]

pixeleyesltd, 2025-01-14:
Thank you for the response and your testing. After further testing on my side it looks like the incorrect door status with no internet was a power supply failing. It just happened to coincide with the ISP outage and a subsequent test. The distances were all over the show.

Re: the UI requiring jQuery, it could be worth instead using the Web Fetch API, getElementById() etc to avoid dependencies. This would make the device a little more future proof if the hosting for dependencies happened to move or go away.

rayshobby, 2025-01-17:
Sure, we will take that into account when rewriting the UI code. Thanks for the suggestion.

closed 2025-01-17 by rayshobby

arfrie22, 2025-01-28:
In the next update, the dependencies will all be stored locally.