OpenGarage/OpenGarage-Firmware #103: Unable to install OpenGarage mobile app from Play Store (Samsung S22 Ultra - Android version 14)

danielbrunt57, 2025-01-17:
[image]

rayshobby, 2025-01-17:
We are fixing this issue at the moment. In the meantime, you can download the app apk directly and install it:
https://apkpure.com/opengarage/com.albahra.opengarage

Alternatively, you can also use our web app, which doesn't require app installation. You can bookmark it to your homescreen (or download the script locally) and it works almost like a native app:
https://raysfiles.com/og/og_devices.html

rayshobby, 2025-01-17:
I am closing this issue for now because this is an App issue, it's not a firmware issue. If you want to continue open it, please submit it at the OpenGarage-App repo:
https://github.com/OpenGarage/OpenGarage-App

closed 2025-01-17 by rayshobby